heleidsn/UAV_Navigation_DRL_AirSim #4: 一些奇怪的报错

ACADIA625379776, 2023-03-29:
第一次编译和运行这个项目，打算尝试SimpleAvoid+Multirotor
在运行thread_train的时候出现了如下报错：
（环境设置如下：
"SimMode": "Multirotor",
  "ViewMode": "SpringArmChase",
  "ClockSpeed": 20,）

Traceback (most recent call last):
  File "D:\UAV_Navigation_DRL_AirSim-main\scripts\utils\thread_train.py", line 220, in run
    model.learn(
  File "D:\UAV_Navigation_DRL_AirSim-main\stable-baselines3\stable_baselines3\sac\sac.py", line 299, in learn
    return super(SAC, self).learn(
  File "D:\UAV_Navigation_DRL_AirSim-main\stable-baselines3\stable_baselines3\common\off_policy_algorithm.py", line 340, in learn
    total_timesteps, callback = self._setup_learn(
  File "D:\UAV_Navigation_DRL_AirSim-main\stable-baselines3\stable_baselines3\common\off_policy_algorithm.py", line 316, in _setup_learn
    return super()._setup_learn(
  File "D:\UAV_Navigation_DRL_AirSim-main\stable-baselines3\stable_baselines3\common\base_class.py", line 435, in _setup_learn
    self._last_obs = self.env.reset()  # pytype: disable=annotation-type-mismatch
  File "D:\UAV_Navigation_DRL_AirSim-main\stable-baselines3\stable_baselines3\common\vec_env\dummy_vec_env.py", line 62, in reset
    self._save_obs(env_idx, obs)
  File "D:\UAV_Navigation_DRL_AirSim-main\stable-baselines3\stable_baselines3\common\vec_env\dummy_vec_env.py", line 92, in _save_obs
    self.buf_obs[key][env_idx] = obs
ValueError: could not broadcast input array from shape (1,7) into shape (1,27)

请问这种情况该怎么做？

ACADIA625379776, 2023-03-29:
使用的config是configs_new/config_Maze_SimpleMultirotor_2D.ini

heleidsn, 2023-04-03:
你好，需要检查一下输入。最近在做一些开发，可能把代码版本弄的比较乱。你上面的问题是obs和网络的输入不一致，尽量看一下get_obs函数的输出和网络输入情况

closed 2023-04-20 by heleidsn

Bruce-Si, 2023-04-22:
同样的问题 请问解决了吗

dilaxy123, 2023-10-24:
> 使用的config是configs_new/config_Maze_SimpleMultirotor_2D.ini

Did you find a solution for this?