heleidsn/UAV_Navigation_DRL_AirSim #8: SimpleAvoid训练卡住

Bruce-Si, 2023-04-22:
您好，训练SimpleAvoid，没过一会无人机就卡在柱子上，无法reset。

[image]

vechi0324, 2023-04-23:
我能问一下你在训练之前是怎样设置的么？我在训练一开始就会卡在柱子上面。

heleidsn, 2023-04-23:
可以贴出一下命令行中的输出看看

Bruce-Si, 2023-04-23:
> 我能问一下你在训练之前是怎样设置的么？我在训练一开始就会卡在柱子上面。

这是我瞎改的配置文件，别的没有设置了。
[options]
project_name = FlappingWing
env_name = SimpleAvoid
dynamic_name = SimpleMultirotor

navigation_3d = False
using_velocity_state = False
reward_type = reward_final

;depth, lgmd, vector
perception = vector

algo = SAC
total_timesteps = 100000

policy_name = mlp
net_arch = [64, 32, 16]
activation_function = tanh
;cnn_feature_num = 25
cnn_feature_num = 5

keyboard_debug = False
generate_q_map = True
q_map_save_steps = 10000

use_wandb = False
wandb_run_name = Maze-2D-mlp-tanh-M3
notes = no

;use_wandb = True
;wandb_run_name = Maze-2D-mlp-tanh-M3
;notes = test

;[wandb]
;name = Maze-2D-mlp-tanh-M3
;notes = test

[environment]
max_depth_meters = 15
screen_height = 60
screen_width = 90

crash_distance = 2
accept_radius = 2

[multirotor]
dt = 0.1
acc_xy_max = 2.0
v_xy_max = 5
v_xy_min = 0.5
v_z_max = 2.0 
yaw_rate_max_deg = 30.0

; configs for DRL algorithms
[DRL]
gamma = 0.99
learning_rate = 1e-3
learning_starts = 2000
buffer_size = 50000
batch_size = 512
train_freq = 100
gradient_steps = 100
action_noise_sigma = 0.1
```
```

Bruce-Si, 2023-04-23:
> 可以贴出一下命令行中的输出看看
这是命令行打印的有信息，请帮看看
[image]

vechi0324, 2023-04-25:
你可以尝试一下修改C盘路径中airsim的settings.json，把仓库路径中的settings_simpledynamics.json内容贴过去试一下，好像这样就能把image_obs尺寸设置成100*80，那个函数就不会出问题了

closed 2023-05-11 by Bruce-Si