heleidsn/UAV_Navigation_DRL_AirSim #9: 运行start_train_with_plot.py出现以下问题，请问应该如何解决呢？

vechi0324, 2023-04-23:
[image]
打开UE4文件后，运行.py文件，就会卡在柱子上，然后命令行中报错如下
[image]
[image]

Bruce-Si, 2023-04-23:
这个是airsim_env.py中的get_obs_vector函数切割array的时候维度不对，代码中把256维的array切成5份，没法平均分就会报错。我是强行把array维度变成5的整倍数，这样代码能跑，但是没一会无人机也会卡住，不知道什么原因。需要作者大佬帮忙

vechi0324, 2023-04-24:
> 这个是airsim_env.py中的get_obs_vector函数切割array的时候维度不对，代码中把256维的array切成5份，没法平均分就会报错。我是强行把array维度变成5的整倍数，这样代码能跑，但是没一会无人机也会卡住，不知道什么原因。需要作者大佬帮忙

由于以前对强化学习和图像这方面确实不太了解，也不是做这个方向的。我现在这边其实是想做一些针对智能agent的测试，想通过airsim的API看能不能跑出一些数据来用用，测试一下。
我看logs_save路径下有保存的模型，就想问一下有什么办法可以直接把作者大佬存的模型运行起来么？就因为我可能也不太需要自己训练DRL这个过程。

closed 2023-04-26 by vechi0324

Ganjiaxiang, 2023-09-22:
> 这个是airsim_env.py中的get_obs_vector函数切割array的时候维度不对，代码中把256维的array切成5份，没法平均分就会报错。我是强行把array维度变成5的整倍数，这样代码能跑，但是没一会无人机也会卡住，不知道什么原因。需要作者大佬帮忙

请问您最后解决了吗？这个问题也困扰我好久了，希望可以得到您的帮助，谢谢。

Bruce-Si, 2023-09-25:
> > 这个是airsim_env.py中的get_obs_vector函数切割array的时候维度不对，代码中把256维的array切成5份，没法平均分就会报错。我是强行把array维度变成5的整倍数，这样代码能跑，但是没一会无人机也会卡住，不知道什么原因。需要作者大佬帮忙
> 
> 请问您最后解决了吗？这个问题也困扰我好久了，希望可以得到您的帮助，谢谢。
后面我觉得用一维vector做输入太粗糙，改成了深度图就没这个卡住的问题了，你可以试试。

Ganjiaxiang, 2023-09-25:
> > > 这个是airsim_env.py中的get_obs_vector函数切割array的时候维度不对，代码中把256维的array切成5份，没法平均分就会报错。我是强行把array维度变成5的整倍数，这样代码能跑，但是没一会无人机也会卡住，不知道什么原因。需要作者大佬帮忙
> > 
> > 
> > 请问您最后解决了吗？这个问题也困扰我好久了，希望可以得到您的帮助，谢谢。
> 
> 后面我觉得用一维vector做输入太粗糙，改成了深度图就没这个卡住的问题了，你可以试试。

好的，感谢您的帮助，谢谢。

wagh311, 2023-12-28:
> > > 这个是airsim_env.py中的get_obs_vector函数切割array的时候维度不对，代码中把256维的array切成5份，没法平均分就会报错。我是强行把array维度变成5的整倍数，这样代码能跑，但是没一会无人机也会卡住，不知道什么原因。需要作者大佬帮忙
> > 
> > 
> > 请问您最后解决了吗？这个问题也困扰我好久了，希望可以得到您的帮助，谢谢。
> 
> 后面我觉得用一维vector做输入太粗糙，改成了深度图就没这个卡住的问题了，你可以试试。

请问除了在配置里改成深度图，还需要在配置中改哪里吗？我把perception改成了depth，但无人机还是在训练一开始就卡住了