heleidsn/UAV_Navigation_DRL_AirSim #11: A top-down view of a custom map

bing-zhub, 2023-05-11:
Can you tell me how to generate a top-down view of a custom map and align the scale and coordinate system? For example, `resources\env_maps\simple_world_light_1.png`.
Also, I would like to know what hardware you trained the model under and how long it took to converge. Thank you.

bing-zhub, 2023-05-11:
I saw you mentioned your thesis in the issue https://github.com/heleidsn/UAV_Navigation_DRL_AirSim/issues/3#issuecomment-1516426451. If possible, I would like to take a look at your thesis . My email address is [bing.zhub@gmail.com](mailto:bing.zhub@gmail.com). Thank you.

heleidsn, 2023-05-15:
Hi,

The top view is generated in UE4. You can change the view mode for your custom UE4 project. But for the published UE4 environment, the only way is to move the camera position on the top, which is not exactly orthographical. 

In conclusion, you can only generate a fine top-down view in your own custom UE4 project. 

As for the thesis, it's under blinded review now. But you can pull up your questions here or via email.

closed 2023-07-25 by heleidsn