heleidsn/UAV_Navigation_DRL_AirSim #13: 您好，这个项目是论文Explainable Deep Reinforcement Learning for UAV autonomous path planning的实现吗？

DRgentle, 2023-05-13:


heleidsn, 2023-05-15:
你好，这个项目主要偏向于模型训练，关于模型可解释性的代码并不在其中

DRgentle, 2023-05-19:
您好，在使用evaluate评估以前的历史模型时遇到问题：
读取：
r"C:\Users\UAV_Navigation_DRL_AirSim-main\logs_eval\SimpleAvoid\sac-reawrd-1\2022_12_26_16_20_SimpleMultirotor_mlp_SAC中的model和config后报错如下：
Traceback (most recent call last):
  File "C:\Users\DRgentle\Desktop\毕设\UAV_Navigation_DRL_AirSim-main\scripts\utils\thread_evaluation.py", line 79, in run
    return self.run_drl_model()
  File "C:\Users\DRgentle\Desktop\毕设\UAV_Navigation_DRL_AirSim-main\scripts\utils\thread_evaluation.py", line 113, in run_drl_model
    unscaled_action, _ = model.predict(obs, deterministic=True)
  File "C:\Users\DRgentle\Desktop\毕设\UAV_Navigation_DRL_AirSim-main\stable-baselines3\stable_baselines3\common\base_class.py", line 573, in predict
    return self.policy.predict(observation, state, episode_start, deterministic)
  File "C:\Users\DRgentle\Desktop\毕设\UAV_Navigation_DRL_AirSim-main\stable-baselines3\stable_baselines3\common\policies.py", line 337, in predict
    observation, vectorized_env = self.obs_to_tensor(observation)
  File "C:\Users\DRgentle\Desktop\毕设\UAV_Navigation_DRL_AirSim-main\stable-baselines3\stable_baselines3\common\policies.py", line 252, in obs_to_tensor
    vectorized_env = is_vectorized_observation(observation, self.observation_space)
  File "C:\Users\DRgentle\Desktop\毕设\UAV_Navigation_DRL_AirSim-main\stable-baselines3\stable_baselines3\common\utils.py", line 375, in is_vectorized_observation
    return is_vec_obs_func(observation, observation_space)
  File "C:\Users\DRgentle\Desktop\毕设\UAV_Navigation_DRL_AirSim-main\stable-baselines3\stable_baselines3\common\utils.py", line 242, in is_vectorized_box_observation
    raise ValueError(
**ValueError: Error: Unexpected observation shape (1, 7) for Box environment, please use (1, 27) or (n_env, 1, 27) for the observation shape.**
似乎时观测空间的尺寸出错，麻烦请问一下，这个项目可以看过去模型的训练效果吗？是否是以前训练模型的时候修改过状态空间吗？

heleidsn, 2023-07-25:
你好，这个项目是我自己使用的，所以确实还存在一些bug，建议自己训练一个简单一点的模型然后再evaluate。

上面这个报错确实是观测空间的问题，可能和配置文件有关，也可能是后面修改了一些代码导致以前的模型不兼容了

closed 2023-09-21 by heleidsn