heleidsn/UAV_Navigation_DRL_AirSim #16: 关于get_obs_image中的图片尺度化

Bruce-Si, 2023-06-14:
请问`get_obs_image`中为什么要用`image_scaled = 255 - image_scaled`

```
    def get_obs_image(self):
        # Normal mode: get depth image then transfer to matrix with state
        # 1. get current depth image and transfer to 0-255  0-20m 255-0m
        image = self.get_depth_image()  # 0-6550400.0 float 32
        image_resize = cv2.resize(image, (self.screen_width,
                                          self.screen_height))
        self.min_distance_to_obstacles = image.min()
        # switch 0 and 255
        image_scaled = np.clip(
            image_resize, 0, self.max_depth_meters) / self.max_depth_meters * 255
        image_scaled = 255 - image_scaled   # ？
        image_uint8 = image_scaled.astype(np.uint8)

        # 2. get current state (relative_pose, velocity)
        state_feature_array = np.zeros((self.screen_height, self.screen_width))
        state_feature = self.dynamic_model._get_state_feature()
        state_feature_array[0, 0:self.state_feature_length] = state_feature

        # 3. generate image with state
        image_with_state = np.array([image_uint8, state_feature_array])
        image_with_state = image_with_state.swapaxes(0, 2)
        image_with_state = image_with_state.swapaxes(0, 1)

        return image_with_state
```

wagh311, 2023-06-19:
借楼还想请问一下，为什么get_depth_image中，depth_img要乘以100呢？
`def get_depth_image(self):

        responses = self.client.simGetImages([
            airsim.ImageRequest("0", airsim.ImageType.DepthVis, True)
        ])

        # check observation
        while responses[0].width == 0:
            print("get_image_fail...")
            responses = self.client.simGetImages(
                airsim.ImageRequest("0", airsim.ImageType.DepthVis, True))

        depth_img = airsim.list_to_2d_float_array(
            responses[0].image_data_float, responses[0].width,
            responses[0].height)

        depth_meter = depth_img * 100

        return depth_meter`

Bruce-Si, 2023-06-20:
> 借楼还想请问一下，为什么get_depth_image中，depth_img要乘以100呢？

这是为了把深度图的距离单位换成米。

wagh311, 2023-06-20:
> > 借楼还想请问一下，为什么get_depth_image中，depth_img要乘以100呢？
> 
> 这是为了把深度图的距离单位换成米。

原来Airsim中深度图数据的单位是厘米啊，明白啦。谢谢解答~~

heleidsn, 2023-06-25:
> 请问`get_obs_image`中为什么要用`image_scaled = 255 - image_scaled`？无人机的状态量，比如相对终点的距离直接归一化到0~1之间，为什么深度图要被255减掉之后再归一化呢?
> 
> ```
>     def get_obs_image(self):
>         # Normal mode: get depth image then transfer to matrix with state
>         # 1. get current depth image and transfer to 0-255  0-20m 255-0m
>         image = self.get_depth_image()  # 0-6550400.0 float 32
>         image_resize = cv2.resize(image, (self.screen_width,
>                                           self.screen_height))
>         self.min_distance_to_obstacles = image.min()
>         # switch 0 and 255
>         image_scaled = np.clip(
>             image_resize, 0, self.max_depth_meters) / self.max_depth_meters * 255
>         image_scaled = 255 - image_scaled   # ？
>         image_uint8 = image_scaled.astype(np.uint8)
> 
>         # 2. get current state (relative_pose, velocity)
>         state_feature_array = np.zeros((self.screen_height, self.screen_width))
>         state_feature = self.dynamic_model._get_state_feature()
>         state_feature_array[0, 0:self.state_feature_length] = state_feature
> 
>         # 3. generate image with state
>         image_with_state = np.array([image_uint8, state_feature_array])
>         image_with_state = image_with_state.swapaxes(0, 2)
>         image_with_state = image_with_state.swapaxes(0, 1)
> 
>         return image_with_state
> ```

你好，这个是因为`image_resize `是障碍物的距离，在归一化到0-255时，距离远的位置数值越接近255，相反，有障碍物的地方反而是0，所以为了让网络更加平衡，在归一化之前用255减了一下，也就是正常没有障碍物的情况下是0，有了障碍物数值才开始增加

Bruce-Si, 2023-06-26:
感谢大佬的trick!

closed 2023-06-26 by Bruce-Si