heleidsn/UAV_Navigation_DRL_AirSim #17: 模型训练的问题

Bruce-Si, 2023-06-29:
请问遇到过下图的问题吗？模型训练的过程中成功率先逐渐上升到0.95左右，后面接着训练成功率又掉下去了。算法是SAC，神经网络是NoCNN。

[image]

heleidsn, 2023-07-03:
哪个环境呢？这种情况之前有的，说明其实reward设计还不够strong，所以在policy更新的时候，一旦跑偏就不太好纠正回来。还有一种可能就是学习率太大了，一次更新太多破坏了原来的policy网络，然后就又相当于重新开始了。

Bruce-Si, 2023-07-03:
谢谢，那大概率是奖励设置的问题，我再调调参。环境是自己建的二维场景，就几根柱子。我想让环境从易到难，逐步训练模型，否则直接在复杂环境中训练始终无法收敛。😞

heleidsn, 2023-07-04:
按照描述这种场景用no_cnn应该会挺快收敛的，可以看一下主要失败原因是碰撞还是到达目标之后转圈，如果是转圈的话可以适当增加目标位置的可接受半径

Bruce-Si, 2023-07-04:
感谢，上面的简单环境后面收敛了，然后换了更复杂点的二维环境：
[image]
发现小障碍物（直径0.5m）难以避开。用的深度图尺寸是200*160，No_CNN下采样到20*20再拉平特征。后面是否要换更大的输入尺寸并且减少下采样呢，我看文章里没有用大尺寸输入的，觉得有点奇怪。

Bruce-Si, 2023-07-04:
另外，再请教个问题，为什么AirSim设置了10倍加速，但实际智能加速到2倍左右，显卡只用了30%。
[image]

heleidsn, 2023-07-04:
小障碍物应该是特征的问题，只要下采样之后还能够看到障碍物即可，讲道理都是可以避开的，你可以研究一下下采样对于小障碍物的影响情况。

关于加速的问题，我也没办法跑满，具体原因没有去细查- -！

Bruce-Si, 2023-07-04:
好的 太小的障碍物下采样肯定就丢了 比如AirSimNH里面的电线或者细杆子，经常撞上去😂

closed 2023-07-19 by Bruce-Si