heleidsn/UAV_Navigation_DRL_AirSim #18: 关于模型设计

Bruce-Si, 2023-07-19:
您好，我用NoCNN，直接将深度图下采样后拉平，再和无人机状态量拼接，再送入全连接层得到动作输出，这样没问题，模型能学会避障和导航。
但是当我减少深度图的下采样倍数，导致拉平后特征维度太大，直接和状态量拼接会使得模型只关注深度图特征，从而只能学会避障而完全不会导航，见下图：
[image]
我想着拼接之前先用全连接层降低深度图特征维度，或者先用全连接层提高状态量特征维度，但是怎么尝试都没有效果。降低深度图特征维度使得模型连避障都不会了，提高状态量维度依然学不会导航。
大佬能给些建议么？

heleidsn, 2023-07-25:
这个就属于特征工程了，我测试过深度特征25+状态特征3，能够实现导航+避障。还有就是奖励函数设计，加大导航奖励的比重这样。除了以上两个方法目前还没有想到其他啥好的解决方案

Bruce-Si, 2023-07-27:
大佬 你之前说过深度图做输入的话CNN在pytorch中容易特征消失 不易收敛 后面有找到原因么

heleidsn, 2023-07-28:
> 大佬 你之前说过深度图做输入的话CNN在pytorch中容易特征消失 不易收敛 后面有找到原因么

并没有，你可以探索一下，我后面就没有再做更多的训练了

closed 2023-07-31 by Bruce-Si

zyian0505, 2023-09-21:
@Bruce-Si 我想请问你是如何使用SAC和NO_CNN的，我在训练一端时间后会出现IndexError: too many indices for tensor of dimension 3问题，是NO_CNN网络结构和我的observations shape不一致的问题。我想请教一下该怎么修改，我个人对airism_env的get_obs_vector(self)函数进行了修改，把split_col = 5改成split_col = 25了，这样在broadcast的时候就可以形状匹配（1，28）了。对于NO_CNN我没有进行修改，以下是我的配置文件[options]
env_name = NH_center
dynamic_name = Multirotor

navigation_3d = True
using_velocity_state = False
reward_type = reward_final

;depth, lgmd, vector
perception = vector

algo = SAC
total_timesteps = 200000

policy_name = No_CNN
net_arch = [128, 64, 32 ,16]
activation_function = tanh
cnn_feature_num = 25

keyboard_debug = False
generate_q_map = True
q_map_save_steps = 5000

use_wandb = False

;use_wandb = True
;wandb_run_name = Maze-2D-mlp-tanh-M3
;notes = test

[wandb]
name = Maze-3D-No_CNN-tanh-M3
notes = test

[environment]
max_depth_meters = 20
screen_height = 80
screen_width = 100

crash_distance = 2
accept_radius = 2

[multirotor]
dt = 0.1
acc_xy_max = 2.0
v_xy_max = 5
v_xy_min = 0.5
v_z_max = 2.0
yaw_rate_max_deg = 30.0

; configs for DRL algorithms
[DRL]
gamma = 0.99
learning_rate = 1e-3
learning_starts = 1000
buffer_size = 50000
batch_size = 128
train_freq = 100
gradient_steps = 100
action_noise_sigma = 0.1

Bruce-Si, 2023-09-25:
可以看下No_CNN模型结构，它的输入是深度图，你需要把配置文件中的`perception = vector`改成`perception = depth`。

zyian0505, 2023-09-25:
@Bruce-Si 请问一下，我的配置文件改成depth之后，会出现**actor为空**的问题该怎么解决呢，就算我用SAC和mlp的默认配置也会出现该问题
配置文件如下：
[options]
env_name = SimpleAvoid
dynamic_name = Multirotor
navigation_3d = False
using_velocity_state = False
reward_type = reward

perception = depth

algo = SAC
total_timesteps = 300000
policy_name = No_CNN
net_arch = [64, 32]
cnn_feature_num = 25
activation_function = tanh

keyboard_debug = False
generate_q_map = True
q_map_save_steps = 5000
use_wandb = False

[wandb]
name = Maze-3D-No_CNN-tanh-M3
notes = test

[environment]
max_depth_meters = 20
screen_height = 60
screen_width = 80

crash_distance = 2
accept_radius = 2

[multirotor]
dt = 0.1
acc_xy_max = 2.0
v_xy_max = 5.0
v_xy_min = 0.5
v_z_max = 2.0
yaw_rate_max_deg = 50.0

crash_distance = 2
accept_radius = 2

[DRL]
gamma = 0.99
learning_rate = 1e-3
learning_starts = 1000
buffer_size = 50000
batch_size = 128
train_freq = 100
gradient_steps = 100
action_noise_sigma = 0.1

错误信息如下：
(60, 80, 2)
Reward type:  reward
run training thread
algo:  SAC
C:\Users\Administrator\AppData\Local\anaconda3\envs\airsim\lib\site-packages\gym\spaces\box.py:73: UserWarning: WARN: Box bound precision lowered by casting to float32
  logger.warn(
Using cuda device
Wrapping the env with a `Monitor` wrapper
Wrapping the env in a DummyVecEnv.
Wrapping the env in a VecTransposeImage.
start training model
Logging to logs_new/SimpleAvoid/2023_09_25_11_59_Multirotor_No_CNN_SAC/tb_logs\SAC_1
Traceback (most recent call last):
  File "D:\UAV_Navigation_DRL_AirSim\scripts\utils\thread_train.py", line 260, in run
    model.learn(total_timesteps)
  File "D:\UAV_Navigation_DRL_AirSim\stable-baselines3\stable_baselines3\sac\sac.py", line 299, in learn
    return super(SAC, self).learn(
  File "D:\UAV_Navigation_DRL_AirSim\stable-baselines3\stable_baselines3\common\off_policy_algorithm.py", line 354, in learn
    rollout = self.collect_rollouts(
  File "D:\UAV_Navigation_DRL_AirSim\stable-baselines3\stable_baselines3\common\off_policy_algorithm.py", line 591, in collect_rollouts
    new_obs, rewards, dones, infos = env.step(actions)
  File "D:\UAV_Navigation_DRL_AirSim\stable-baselines3\stable_baselines3\common\vec_env\base_vec_env.py", line 162, in step
    return self.step_wait()
  File "D:\UAV_Navigation_DRL_AirSim\stable-baselines3\stable_baselines3\common\vec_env\vec_transpose.py", line 95, in step_wait
    observations, rewards, dones, infos = self.venv.step_wait()
  File "D:\UAV_Navigation_DRL_AirSim\stable-baselines3\stable_baselines3\common\vec_env\dummy_vec_env.py", line 43, in step_wait
    obs, self.buf_rews[env_idx], self.buf_dones[env_idx], self.buf_infos[env_idx] = self.envs[env_idx].step(
  File "D:\UAV_Navigation_DRL_AirSim\stable-baselines3\stable_baselines3\common\monitor.py", line 90, in step
    observation, reward, done, info = self.env.step(action)
  File "C:\Users\Administrator\AppData\Local\anaconda3\envs\airsim\lib\site-packages\gym\wrappers\order_enforcing.py", line 11, in step
    observation, reward, done, info = self.env.step(action)
  File "D:\UAV_Navigation_DRL_AirSim\gym_env\gym_env\envs\airsim_env.py", line 260, in step
    self.print_train_info_airsim(action, obs, reward, info)
  File "D:\UAV_Navigation_DRL_AirSim\gym_env\gym_env\envs\airsim_env.py", line 942, in print_train_info_airsim
    feature_all = self.model.actor.features_extractor.feature_all
AttributeError: 'NoneType' object has no attribute 'actor'

Bruce-Si, 2023-09-25:
@zyian0505 qq发你邮箱了

MiraiMoe, 2024-01-26:
作者代码里写的获得feature的方法是feature_all = self.model.actor.features_extractor.feature_all，但是model初始赋值为None，所以报错，可以将feature取值改为get_obs_image函数的返回值image_with_state，此时在depth模式下训练可以进行