heleidsn/UAV_Navigation_DRL_AirSim #23: The UAV lands randomly in the middle of the iteration

didzdissanayaka8, 2023-10-24:
[image]

I realized that the UAV lands randomly in the middle of the current iteration and it starts from there again in the next iteration. Is there a configuration time parameter I need to modify to avoid that?  

Does it affect the model training or it just continues from there onwards.

abiaoaaaaa, 2024-01-07:
Hello, why doesn't the gui interface draw any curves after I run it? Can you explain your configuration process?

SuperheroZLP, 2024-10-27:
hello，do you know why it is：

D:\software\Anaconda\Anaconda\envs\DRL\python.exe D:\software\AirSim\UAV_Navigation_DRL_AirSim\scripts\start_evaluate_with_plot.py
init training thread
init airsim-gym-env.
Traceback (most recent call last):
File "D:\software\AirSim\UAV_Navigation_DRL_AirSim\scripts\start_evaluate_with_plot.py", line 64, in
main()
File "D:\software\AirSim\UAV_Navigation_DRL_AirSim\scripts\start_evaluate_with_plot.py", line 42, in main
evaluate_thread = EvaluateThread(
File "D:\software\AirSim\UAV_Navigation_DRL_AirSim\scripts\utils\thread_evaluation.py", line 65, in init
self.env = gym.make('airsim-env-v0')
File "D:\software\Anaconda\Anaconda\envs\DRL\lib\site-packages\gym\envs\registration.py", line 669, in make
env = PassiveEnvChecker(env)
File "D:\software\Anaconda\Anaconda\envs\DRL\lib\site-packages\gym\wrappers\env_checker.py", line 20, in init
assert hasattr(
AssertionError: The environment must specify an action space.

VirtualCoder0, 2025-05-16:
> [image]
> 
> I realized that the UAV lands randomly in the middle of the current iteration and it starts from there again in the next iteration. Is there a configuration time parameter I need to modify to avoid that?
> 
> Does it affect the model training or it just continues from there onwards.

Do you have an answer to explain the question now? I'm also confused about this phenomenon