heleidsn/UAV_Navigation_DRL_AirSim #26: 关于clockspeed设置不同大小的影响

wagh311, 2023-12-04:
我在使用深度强化学习训练多无人机避障时，将 ClockSpeed 设置为 5 来加快训练。当我完成训练后，如果我仍在 ClockSpeed 设置为 5 的环境下进行测试，无人机可以成功避碰。 但是当我将 ClockSpeed 设置为 1 进行测试时，无人机就会发生碰撞。我想请问一下这是为什么啊？难道ClockSpeed 设置会影响无人机本身的动力学吗？为什么在测试时也必须要设置跟训练时一样的ClockSpeed 值才能成功复现训练结果啊？如果是这样的话，那把策略迁移到实际环境中时不会出问题吗？毕竟真实环境下，应该相当于ClockSpeed=1的情况吧？希望大佬帮忙解答一下，非常感谢！

heleidsn, 2024-03-18:
不好意思刚看到，不知道你问题解决了吗，ClockSpeed设置为1之后碰撞概率大吗，可以查看一下网络输出是否正确。ClockSpeed 设置讲道理不会影响到运动学和动力学，需要实际情况分析一下

wagh311, 2024-03-18:
我目前没有还没有找到解决这个问题的方法，但我发现这个问题确实存在。clockspeed参数的设置会对Airsim中的物理参数造成影响，而并非单纯的加速仿真。例如设置clockspeed = 1，在这种情况下，设置一架UAV沿X轴以5m/s的速度运动，那么此时测试UAV的速度，发现确实是5m/s；但是，如果在clockspeed =5时设置一架UAV沿X轴以5m/s的速度运动，那么再去测试UAV的速度，会发现只有大约1m/s，也即5/5 = 1。（真实数据/时钟速度）。这也是为什么我说在ClockSpeed=5的设置下训练好的策略，无法成功的应用于ClockSpeed=1的测试中。
此外，也有其他人发现了这个问题，例如：[https://github.com/microsoft/AirSim/issues/4841](url)
还有一篇论文也提到了这个现象：
[simul_2022_1_90_50041.pdf](https://github.com/heleidsn/UAV_Navigation_DRL_AirSim/files/14630540/simul_2022_1_90_50041.pdf)

heleidsn, 2024-04-16:
> 我目前没有还没有找到解决这个问题的方法，但我发现这个问题确实存在。clockspeed参数的设置会对Airsim中的物理参数造成影响，而并非单纯的加速仿真。例如设置clockspeed = 1，在这种情况下，设置一架UAV沿X轴以5m/s的速度运动，那么此时测试UAV的速度，发现确实是5m/s；但是，如果在clockspeed [...]

非常感谢你发现的这个问题，后面找机会研究和修复一下~

wagh311, 2024-04-17:
嗯嗯，期待您的更新~~

SuperheroZLP, 2024-10-27:
佬，你是怎么跑成功的啊，能不能帮我看看这，一跑就报错：

D:\software\Anaconda\Anaconda\envs\DRL\python.exe D:\software\AirSim\UAV_Navigation_DRL_AirSim\scripts\start_evaluate_with_plot.py
init training thread
init airsim-gym-env.
Traceback (most recent call last):
File "D:\software\AirSim\UAV_Navigation_DRL_AirSim\scripts\start_evaluate_with_plot.py", line 64, in
main()
File "D:\software\AirSim\UAV_Navigation_DRL_AirSim\scripts\start_evaluate_with_plot.py", line 42, in main
evaluate_thread = EvaluateThread(
File "D:\software\AirSim\UAV_Navigation_DRL_AirSim\scripts\utils\thread_evaluation.py", line 65, in init
self.env = gym.make('airsim-env-v0')
File "D:\software\Anaconda\Anaconda\envs\DRL\lib\site-packages\gym\envs\registration.py", line 669, in make
env = PassiveEnvChecker(env)
File "D:\software\Anaconda\Anaconda\envs\DRL\lib\site-packages\gym\wrappers\env_checker.py", line 20, in init
assert hasattr(
AssertionError: The environment must specify an action space.