heleidsn/UAV_Navigation_DRL_AirSim #31: ERROR Internal wandb error: file data was not synced

Guoyulun, 2024-04-06:
很抱歉打扰您，当我运行 `python scripts/start_train_with_plot.py`时，出现了以下错误
`
Thread HandlerThread:wandb.init()...
Traceback (most recent call last):
  File "D:\anaconda3\envs\airsim\lib\site-packages\wandb\sdk\internal\internal_util.py", line 49, in run
    self._run()
  File "D:\anaconda3\envs\airsim\lib\site-packages\wandb\sdk\internal\internal_util.py", line 100, in _run
    self._process(record)
  File "D:\anaconda3\envs\airsim\lib\site-packages\wandb\sdk\internal\internal.py", line 279, in _process
    self._hm.handle(record)
  File "D:\anaconda3\envs\airsim\lib\site-packages\wandb\sdk\internal\handler.py", line 138, in handle
    handler(record)
  File "D:\anaconda3\envs\airsim\lib\site-packages\wandb\sdk\internal\handler.py", line 148, in handle_request
    handler(record)
  File "D:\anaconda3\envs\airsim\lib\site-packages\wandb\sdk\internal\handler.py", line 686, in handle_request_run_start
    self._system_monitor.probe(publish=True)
  File "D:\anaconda3\envs\airsim\lib\site-packages\wandb\sdk\internal\system\system_monitor.py", line 228, in probe     
    self.system_info.publish(system_info)
  File "D:\anaconda3\envs\airsim\lib\site-packages\wandb\sdk\internal\system\system_info.py", line 230, in publish      
    self._save_patches()
  File "D:\anaconda3\envs\airsim\lib\site-packages\wandb\sdk\internal\system\system_info.py", line 93, in _save_patches 
    subprocess.check_call(
  File "D:\anaconda3\envs\airsim\lib\subprocess.py", line 359, in check_call
    retcode = call(*popenargs, **kwargs)
  File "D:\anaconda3\envs\airsim\lib\subprocess.py", line 340, in call
    with Popen(*popenargs, **kwargs) as p:
  File "D:\anaconda3\envs\airsim\lib\subprocess.py", line 858, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "D:\anaconda3\envs\airsim\lib\subprocess.py", line 1327, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
NotADirectoryError: [WinError 267] 目录名称无效。
wandb: ERROR Internal wandb error: file data was not synced
Problem at: C:\Users\A1\Desktop\AirsimProject\UAV_Navigation_DRL_AirSim\scripts\utils\thread_train.py 69 __init__
wandb: ERROR transport failed
Traceback (most recent call last):
  File "scripts/start_train_with_plot.py", line 54, in <module>
    main()
  File "scripts/start_train_with_plot.py", line 36, in main
    training_thread = TrainingThread(config_file)
  File "C:\Users\A1\Desktop\AirsimProject\UAV_Navigation_DRL_AirSim\scripts\utils\thread_train.py", line 69, in __init__
    wandb.init(
  File "D:\anaconda3\envs\airsim\lib\site-packages\wandb\sdk\wandb_init.py", line 1206, in init
    raise e
  File "D:\anaconda3\envs\airsim\lib\site-packages\wandb\sdk\wandb_init.py", line 1187, in init
    run = wi.init()
  File "D:\anaconda3\envs\airsim\lib\site-packages\wandb\sdk\wandb_init.py", line 818, in init
    run_start_result = run_start_handle.wait(timeout=30)
  File "D:\anaconda3\envs\airsim\lib\site-packages\wandb\sdk\lib\mailbox.py", line 281, in wait
    raise MailboxError("transport failed")
wandb.sdk.lib.mailbox.MailboxError: transport failed
Traceback (most recent call last):
  File "scripts/start_train_with_plot.py", line 54, in <module>
    main()
  File "scripts/start_train_with_plot.py", line 36, in main
    training_thread = TrainingThread(config_file)
  File "C:\Users\A1\Desktop\AirsimProject\UAV_Navigation_DRL_AirSim\scripts\utils\thread_train.py", line 69, in __init__
    wandb.init(
  File "D:\anaconda3\envs\airsim\lib\site-packages\wandb\sdk\wandb_init.py", line 1206, in init
    raise e
  File "D:\anaconda3\envs\airsim\lib\site-packages\wandb\sdk\wandb_init.py", line 1187, in init
    run = wi.init()
  File "D:\anaconda3\envs\airsim\lib\site-packages\wandb\sdk\wandb_init.py", line 818, in init
    run_start_result = run_start_handle.wait(timeout=30)
  File "D:\anaconda3\envs\airsim\lib\site-packages\wandb\sdk\lib\mailbox.py", line 281, in wait
    raise MailboxError("transport failed")
wandb.sdk.lib.mailbox.MailboxError: transport failed
wandb: While tearing down the service manager. The following error has occurred: [WinError 10054] 远程主机强迫关闭了一个现有的连接。
`
看起来似乎是因为目录错误，请问您有办法解决吗？
谢谢！

heleidsn, 2024-04-06:
你好，这个感觉像是wandb的问题，你可以在config文件里面关闭wandb再试试

Guoyulun, 2024-04-07:
非常感谢您的回复，wandb的问题我已经解决了，但是我运行后又出现了错误：
`Traceback (most recent call last):
  File "C:\Users\A1\Desktop\AirsimProject\UAV_Navigation_DRL_AirSim\scripts\utils\thread_train.py", line 231, in run
    model.learn(total_timesteps)
  File "c:\users\a1\desktop\airsimproject\uav_navigation_drl_airsim\stable-baselines3\stable_baselines3\td3\td3.py", line 244, in learn
    return super(TD3, self).learn(
  File "c:\users\a1\desktop\airsimproject\uav_navigation_drl_airsim\stable-baselines3\stable_baselines3\common\off_policy_algorithm.py", line 340, in learn       
    total_timesteps, callback = self._setup_learn(
  File "c:\users\a1\desktop\airsimproject\uav_navigation_drl_airsim\stable-baselines3\stable_baselines3\common\off_policy_algorithm.py", line 316, in _setup_learn
    return super()._setup_learn(
  File "c:\users\a1\desktop\airsimproject\uav_navigation_drl_airsim\stable-baselines3\stable_baselines3\common\base_class.py", line 435, in _setup_learn
    self._last_obs = self.env.reset()  # pytype: disable=annotation-type-mismatch
  File "c:\users\a1\desktop\airsimproject\uav_navigation_drl_airsim\stable-baselines3\stable_baselines3\common\vec_env\dummy_vec_env.py", line 62, in reset       
    self._save_obs(env_idx, obs)
  File "c:\users\a1\desktop\airsimproject\uav_navigation_drl_airsim\stable-baselines3\stable_baselines3\common\vec_env\dummy_vec_env.py", line 92, in _save_obs
    self.buf_obs[key][env_idx] = obs
ValueError: could not broadcast input array from shape (1,8) into shape (1,28)`
我查看了stable_baseline3中的代码，但是不是很理解obs的shape为什么是(1,8)，请问您有遇到过相似的问题吗？
（我是刚接触无人机路径规划的小白，问题有点多，实在不好意思）

Guoyulun, 2024-04-08:
我已经解决了，cnn_feature_num = 5就可以了

感谢您的代码，现在可以正常工作了

closed 2024-04-08 by Guoyulun