heleidsn/UAV_Navigation_DRL_AirSim #32: Training a model

1234238, 2024-04-09:
Hello, we have recently been training a drone obstacle avoidance model, and I am using the "simple avoid" scenario. However, the training results are not very good with the default settings. I wonder if you have any training tips? Additionally, it seems that the model_save and other models cannot be loaded. Are there any other models provided? Thank you.

heleidsn, 2024-04-10:
Could you please share the training curve such as episode reward or success rate? The training resutlt is much related with the network type. NO_CNN net will be easier for training.

1234238, 2024-04-10:
The model you are using is Multirotor, and the ClockSpeed is set to 5

This is my training configuration:
--------------------------------------------------------
[options]
env_name = SimpleAvoid
dynamic_name = Multirotor
;dynamic_name = SimpleMultirotor

navigation_3d = False
using_velocity_state = False
reward_type = reward_final

;depth, lgmd, vector
perception = depth

algo = SAC
total_timesteps = 500000

;policy_name = mlp
policy_name = CNN_FC
net_arch = [256,256]
activation_function = tanh
cnn_feature_num = 25

keyboard_debug = False
generate_q_map = True
q_map_save_steps = 10000

use_wandb = False

[wandb]
name = Maze-2D-mlp-tanh-M3
notes = test

[environment]
max_depth_meters = 15
screen_height = 60
screen_width = 90

crash_distance = 1
accept_radius = 4

[multirotor]
dt = 0.2
acc_xy_max = 2.0
v_xy_max = 5
v_xy_min = 1
v_z_max = 2.0
yaw_rate_max_deg = 30.0

; configs for DRL algorithms
[DRL]
gamma = 0.99
learning_rate = 1e-3
learning_starts = 2000
buffer_size = 50000
batch_size = 512
train_freq = 100
gradient_steps = 100
action_noise_sigma = 0.1

--------------------------------------------------------

Here is my training result.
[image]

--------------------------------------------------------
The issue I'm facing right now is that the drone tends to fly directly towards the target, but the probability of collision is high, making it unable to successfully navigate around obstacles and reach the destination. Do you have any training suggestions? Thank you for your response.

1234238, 2024-04-10:
I also adjusted the coefficients of the rewards. I increased the reward for reaching the destination and reduced the penalty for the drone flying directly towards the target. I hope to teach it obstacle avoidance, but the training results are not very promising. If I reduce the penalty too much for the drone flying directly towards the target, it may end up circling in place. On the other hand, if the penalty is not reduced enough, it fails to effectively avoid obstacles. This is a problem that is puzzling me.

------------------------------------
This is reward function

 def compute_reward_final(self, done, action):
        reward = 0
        reward_reach = 100
        reward_crash = -25
        reward_outside = -10
       
        if self.env_name == 'NH_center':
            distance_reward_coef = 500
        else:
            distance_reward_coef = 50

        if not done:

            # 1 - goal reward
            distance_now = self.get_distance_to_goal_3d()
            reward_distance = distance_reward_coef * (self.previous_distance_from_des_point - distance_now) / \
                self.dynamic_model.goal_distance   # normalized to 100 according to goal_distance
            self.previous_distance_from_des_point = distance_now

            # 2 - Position punishment
            current_pose = self.dynamic_model.get_position()
            goal_pose = self.dynamic_model.goal_position
            x = current_pose[0]
            y = current_pose[1]
            z = current_pose[2]
            x_g = goal_pose[0]
            y_g = goal_pose[1]
            z_g = goal_pose[2]

            punishment_xy = np.clip(self.getDis(
                x, y, 0, 0, x_g, y_g) / 10, 0, 1)
            punishment_z = 0.5 * np.clip((z - z_g)/5, 0, 1)

            punishment_pose = punishment_xy + punishment_z

            if self.min_distance_to_obstacles < 3:
                punishment_obs = 1 - np.clip((self.min_distance_to_obstacles - self.crash_distance) / 5, 0, 1)
            else:
                punishment_obs = 0

            punishment_action = 0

            # add yaw_rate cost
            yaw_speed_cost = abs(action[-1]) / self.dynamic_model.yaw_rate_max_rad

            if self.dynamic_model.navigation_3d:
                # add action and z error cost
                v_z_cost = ((abs(action[1]) / self.dynamic_model.v_z_max)**2)
                z_err_cost = (
                    (abs(self.dynamic_model.state_raw[1]) / self.dynamic_model.max_vertical_difference)**2)
                punishment_action += (v_z_cost + z_err_cost)

            punishment_action += yaw_speed_cost

            yaw_error = self.dynamic_model.state_raw[2]
            yaw_error_cost = abs(yaw_error / 90)

            reward = reward_distance - 0 * punishment_pose - 1 * \
                punishment_obs - 0 * punishment_action - 0.2 * yaw_error_cost

           # print(reward_distance, punishment_obs, punishment_pose, yaw_error_cost)
        else:
            if self.is_in_desired_pose():
                reward = reward_reach

            if self.is_crashed():
                reward = reward_crash

            if self.is_not_inside_workspace():
                reward = reward_outside

        return reward

1234238, 2024-04-10:
Thank you for your suggestion. I use the NO_CNN network and it trains very well. It’s really magical.

1234238, 2024-04-10:
Hello, after I successfully trained SAC using NO_CNN, I assigned the same parameters to the TD3 algorithm, but the strange thing is that when the number of decision steps reaches 2000 (after the pre-processing is completed), the drone will circle in place . I am not dont Understand why this happens

1234238, 2024-04-10:
> Could you please share the training curve such as episode reward or success rate? The training resutlt is much related with the network type. NO_CNN net will be easier for training.

Hello, after I successfully trained SAC using NO_CNN, I assigned the same parameters to the TD3 algorithm, but the strange thing is that when the number of decision steps reaches 2000 (after the pre-processing is completed), the drone will circle in place . I am not dont Understand why this happens

heleidsn, 2024-04-16:
TD3 needs more time to converge. Maybe you can try to increase the random explore timesteps to 5000. Always circle means the agent trapped in the local minima.

1234238, 2024-04-16:
> TD3 needs more time to converge. Maybe you can try to increase the random explore timesteps to 5000. Always circle means the agent trapped in the local minima.

Thank you very much for your reply. After hard work, my model has been trained. Another question is that I want to change the settings of some obstacles in the virtual environment. I don’t know how I should do it?

heleidsn, 2024-04-17:
Do you mean a custom environment? The environment is created using UE4 with AirSim plugin to provide basic API such as UAV control. You can follow this [link](https://microsoft.github.io/AirSim/unreal_custenv/) to create your own environment. Please note your AirSim version as the API maybe a little bit different between different AirSim releases. It will cost some time to study how to do this, or you can download the release version from AirSim repo. 

Good luck!

1234238, 2024-04-22:
> Do you mean a custom environment? The environment is created using UE4 with AirSim plugin to provide basic API such as UAV control. You can follow this [...]

Thank you very much for your answer. I have read the relevant reading, but I want to simply modify the environment, such as some obstacle distribution of Simple_avoid, but Simple_avoid does not seem to have an .unproject file in the file, so I cannot open it with UE4. These seem to It was not mentioned. I wonder if you know about it?

heleidsn, 2024-04-23:
Simple_avoid is a custom environment created by myself. You can generate it from scratch in UE4 and add the AirSim plugin. As it is a long time, I cannot find the unproject file now. Maybe you still have to learn the AirSim plugin and UE4 and build your own environment.

1234238, 2024-04-23:
> Simple_avoid is a custom environment created by myself. You can generate it from scratch in UE4 and add the AirSim plugin. As it is a long time, I cannot find the unproject file now. Maybe you still have to learn the AirSim plugin and UE4 and build your own environment.

Thank you very much for your answer. I have another question about the production of the UI interface. If I want the UI interface to display a map like zhangjiajie, do I only need to make a background image? In addition, can I ask how you make a background image like NH?

heleidsn, 2024-04-23:
For a customed environment, we can use UE4 file to generate a good 2D background image. However, for other environment already built, such as zhangjiajie and NH, you can only get the image while manually move the camera to a particular height to get the top view. There will be some perspective problem but I didn't find a good solution for it. You can type M on the keyboard in the environment and move the camera as this [link](https://microsoft.github.io/AirSim/camera_views/#controlling-manual-camera).

JerryBryant24, 2024-05-24:
Hello, I changed the strategy to No_CNN, but I got an error after running it twice.
[image]

JerryBryant24, 2024-05-24:
[image]

1234238, 2024-05-29:
> Hello, I changed the strategy to No_CNN, but I got an error after running it twice. ![2](https://private-user-images.githubusercontent.com/57161255/333521271-1f904264-c3b4-4121-bb62-39305e611fbc.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTcwMDE4OTUsIm5iZiI6MTcxNzAwMTU5NSwicGF0aCI6Ii81NzE2MTI1NS8zMzM1MjEyNzEtMWY5MDQyNjQtYzNiNC00MTIxLWJiNjItMzkzMDVlNjExZmJjLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MjklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTI5VDE2NTMxNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWYwODAyMGQ0NDJiNjg4YzM0ZGQwMDEyMmZhZGEwNzA2ZWZiODRjOTZlOGFhYzI3NTc5NjMxMWQ2ODIwNzZiNGUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.RYe_NkcU1njStRpfAsva7xyt_KS2i0DEg6BBXg3ZFqk)

I think your parameters might need to be adjusted.

1234238, 2024-05-29:
> For a customed environment, we can use UE4 file to generate a good 2D background image. However, for other environment already built, such as zhangjiajie and NH, you can only get the image while [...]

Hello, I would like to ask if you have any reading recommendations for creating beautiful background images like NH_Center in UE4? I have found the unproject file for Zhangjiajie, or would you consider adding the Zhangjiajie environment to the environments you support? Looking forward to your reply.

jinjikikko976, 2024-08-08:
> > 您好，我将策略更改为 No_CNN，但是运行两次后出现错误。![2](https://private-user-images.githubusercontent.com/57161255/333521271-1f904264-c3b4-4121-bb62-39305e611fbc.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTcwMDE4OTUsIm5iZiI6MTcxNzAwMTU5NSwicGF0aCI6Ii81NzE2MTI1NS8zMzM1MjEyNzEtMWY5MDQyNjQtYzNiNC00MTIxLWJiNjItMzkzMDVlNjExZmJjLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MjklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTI5VDE2NTMxNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWYwODAyMGQ0NDJiNjg4YzM0ZGQwMDEyMmZhZGEwNzA2ZWZiODRjOTZlOGFhYzI3NTc5NjMxMWQ2ODIwNzZiNGUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.RYe_NkcU1njStRpfAsva7xyt_KS2i0DEg6BBXg3ZFqk)
> 
> 我认为您的参数可能需要调整。

您好，请问您可以分享下你的参数配吗

jinjikikko976, 2024-08-08:
> 我认为您的参数可能需要调整。
 您好，请问您可以分享下您的参数配置吗？感谢

leeexiaoyao, 2024-11-03:
> > 你好，我把策略改成了No_CNN，但是运行两次之后就出错了。![2](https://private-user-images.githubusercontent.com/57161255/333521271-1f904264-c3b4-4121-bb62-39305e611fbc.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTcwMDE4OTUsIm5iZiI6MTcxNzAwMTU5NSwicGF0aCI6Ii81NzE2MTI1NS8zMzM1MjEyNzEtMWY5MDQyNjQtYzNiNC00MTIxLWJiNjItMzkzMDVlNjExZmJjLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MjklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTI5VDE2NTMxNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWYwODAyMGQ0NDJiNjg4YzM0ZGQwMDEyMmZhZGEwNzA2ZWZiODRjOTZlOGFhYzI3NTc5NjMxMWQ2ODIwNzZiNGUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.RYe_NkcU1njStRpfAsva7xyt_KS2i0DEg6BBXg3ZFqk)
> 
> 我认为您的参数可能需要调整。

Hello, can you share your config file? Thanks very much!

1-pluto1, 2024-11-25:
> Hello, can you share your config file? Thanks very much!

你解决这个问题了吗？请问可以请教一下吗？

datutu-L, 2024-12-12:
> 您好，我将策略更改为 No_CNN，但运行两次后出现错误。 ![2](https://private-user-images.githubusercontent.com/57161255/333521271-1f904264-c3b4-4121-bb62-39305e611fbc.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzM5OTMyNTgsIm5iZiI6MTczMzk5Mjk1OCwicGF0aCI6Ii81NzE2MTI1NS8zMzM1MjEyNzEtMWY5MDQyNjQtYzNiNC00MTIxLWJiNjItMzkzMDVlNjExZmJjLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMTIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjEyVDA4NDIzOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTJjMmIwNzc0ZmRjYjM3ZDYwYTE1YjY1NzNmMjhhY2IwMmQyNzVmZGRiYjg1MTZmZTM1ZmI5ODU1OGVmYmIwZjkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.k00BLf5JqzpHunkY81M01E2wHqltAikTZgWHqLZ4IL8)

我将策略更改为 No_CNN出现跟你一样的错误，请问你解决这个问题了吗

1-pluto1, 2024-12-17:
> > 你解决这个问题了吗？请问可以请教一下吗？
> 
> 您好，请问您有解决这个问题吗

解决了，需要修改一下no_cnn的池化参数

DangLiao, 2025-04-10:
> > > 你解决这个问题了吗？请问可以请教一下吗？
> > 
> > 
> > 您好，请问您有解决这个问题吗
> 
> 解决了，需要修改一下no_cnn的池化参数

您好，请问该怎么修改no_cnn的池化参数

SakuraJimaMaiiiiiiii, 2025-04-14:
> t

hello , i would like to ask how to adjust parameters in code

SakuraJimaMaiiiiiiii, 2025-04-14:
> I think your parameters might need to be adjusted.

hello , i would like to ask how to adjust parameters in code

SakuraJimaMaiiiiiiii, 2025-04-15:
> > > > > 你解决这个问题了吗？请问可以请教一下吗？
> > > > 
> > > > 
> > > > 您好，请问您有解决这个问题吗
> > > 
> > > 
> > > 解决了，需要修改一下no_cnn的池化参数
> > 
> > 
> > 您好，请问该怎么修改no_cnn的池化参数
> 
> [image]我几个月没有弄这个项目了，主要是这一行，也可能这个文件上面几行也可能改一点，你自己看着比对着改一下能不能运行

你好 我看到你的policy_name是mlp 我换成No_CNN后 把cnn_feature_num换成5后还是报错，我自己没找到在哪里改这里输入维度，麻烦了

[image]

leeexiaoyao, 2025-04-16:
> > > > > > 你解决这个问题了吗？请问可以请教一下吗？
> > > > > 
> > > > > 
> > > > > 您好，请问您有解决这个问题吗
> > > > 
> > > > 
> > > > 解决了，需要修改一下no_cnn的池化参数
> > > 
> > > 
> > > 您好，请问该怎么修改no_cnn的池化参数
> > 
> > 
> > [image]我几个月没有弄这个项目了，主要是这一行，也可能这个文件上面几行也可能改一点，你自己看着比对着改一下能不能运行
> 
> 你好 我看到你的policy_name是mlp 我换成No_CNN后 把cnn_feature_num换成5后还是报错，我自己没找到在哪里改这里输入维度，麻烦了
> 
> [image]
先调整cnn_feature_num，然后不行的话policy可以调成No_CNN ，另外多问问gpt

3055Y, 2025-05-07:
> > > > > > 你解决这个问题了吗？请问可以请教一下吗？
> > > > > 
> > > > > 
> > > > > 您好，请问您有解决这个问题吗
> > > > 
> > > > 
> > > > 解决了，需要修改一下no_cnn的池化参数
> > > 
> > > 
> > > 您好，请问该怎么修改no_cnn的池化参数
> > 
> > 
> > [image]我几个月没有弄这个项目了，主要是这一行，也可能这个文件上面几行也可能改一点，你自己看着比对着改一下能不能运行
> 
> 你好 我看到你的policy_name是mlp 我换成No_CNN后 把cnn_feature_num换成5后还是报错，我自己没找到在哪里改这里输入维度，麻烦了
> 
> [image]

把cnn_feature_num改成12