heleidsn/UAV_Navigation_DRL_AirSim #33: 运行python scripts/start_evaluate_with_plot.py时报错

rad104ct1ve, 2024-04-10:
在训练时可以正常运行，在以logs目录下保存的文件夹为eval path运行evaluate的时候就会报错，报错如下：

D:\anaconda\envs\uavsim\python.exe scripts\start_evaluate_with_plot.py 
Traceback (most recent call last):
  File "scripts\start_evaluate_with_plot.py", line 62, in <module>
    main()
  File "scripts\start_evaluate_with_plot.py", line 36, in main
    gui = TrainingUi(config=config_file)
  File "E:\DRL\UAV_Navigation_DRL_AirSim-main\UAV_Navigation_DRL_AirSim-main\scripts\utils\ui_train.py", line 37, in __init__
    self.init_ui()
  File "E:\DRL\UAV_Navigation_DRL_AirSim-main\UAV_Navigation_DRL_AirSim-main\scripts\utils\ui_train.py", line 59, in init_ui
    self.dynamics = self.cfg.get('options', 'dynamic_name')
  File "D:\anaconda\envs\uavsim\lib\configparser.py", line 781, in get
    d = self._unify_values(section, vars)
  File "D:\anaconda\envs\uavsim\lib\configparser.py", line 1149, in _unify_values
    raise NoSectionError(section) from None
configparser.NoSectionError: No section: 'options'

我将eval_path改成logs_eval\SimpleAvoid\sac-reawrd-1下的文件夹仍会报错：

Traceback (most recent call last):
  File "scripts\start_evaluate_with_plot.py", line 62, in <module>
    main()
  File "scripts\start_evaluate_with_plot.py", line 36, in main
    gui = TrainingUi(config=config_file)
  File "E:\DRL\UAV_Navigation_DRL_AirSim-main\UAV_Navigation_DRL_AirSim-main\scripts\utils\ui_train.py", line 37, in __init__
    self.init_ui()
  File "E:\DRL\UAV_Navigation_DRL_AirSim-main\UAV_Navigation_DRL_AirSim-main\scripts\utils\ui_train.py", line 59, in init_ui
    self.dynamics = self.cfg.get('options', 'dynamic_name')
  File "D:\anaconda\envs\uavsim\lib\configparser.py", line 781, in get
    d = self._unify_values(section, vars)
  File "D:\anaconda\envs\uavsim\lib\configparser.py", line 1149, in _unify_values
    raise NoSectionError(section) from None
configparser.NoSectionError: No section: 'options'

请问大佬这是什么原因啊，本人强化学习小白，希望大佬解答一下，非常感谢！

heleidsn, 2024-04-16:
这是config文件的问题，因为代码进行了多次更新，所以之前的config文件里面有些命名就出现了错误，比如`No section: 'options'`的问题，你在验证的时候尽量使用自己训练得到的policy会好一些

SuperheroZLP, 2024-10-23:
你好，请问你解决这个问题了吗，我也遇到了这个问题

Z-Frankie, 2025-01-11:
我将里面的所有相对路径改为绝对路径就可以了。