heleidsn/UAV_Navigation_DRL_AirSim #36: 环境切换

JerryBryant24, 2024-05-24:
当我把简单环境切换为NH时出现了下面的错误
[image]
config为：config_NH_center_SimpleMultirotor_3D.ini
 期待您的解答

heleidsn, 2024-05-25:
这个应该是和设置中的percetpition有关，也就是网络的输入，可以具体排查一下设置和get_state的部分

JerryBryant24, 2024-05-27:
Hello, I changed the strategy to No_CNN, but I got an error after running it twice.
[image]

heleidsn, 2024-05-28:
It‘s still because dimensions of the state do not match the neural network

MalayPhadke, 2024-08-23:
Hello, I was having the same error. setting perception to depth instead of vector in the config file makes it go away. I don't know why. Thank you!

Flieluo, 2025-09-19:
> 您好，我遇到了同样的错误。在配置文件中将感知设置为深度而不是矢量会使其消失。我不知道为什么。谢谢！

Could you share the parameter file compatible with the NH 3D environment?