heleidsn/UAV_Navigation_DRL_AirSim #37: What is different between  train.py and start_train_with_plot.py?

khosrooo, 2024-06-08:
which one is main code?

heleidsn, 2024-06-09:
Hi, if I remember correctly, both of them can be used to start the training process. The difference is whether you want to turn on a plot for visualization during training.

khosrooo, 2024-06-09:
Thank u Sir
i have a question 
I want to propose a deep reinforcement learning-based controller for a quadcopter that tracks ground objects. The reward function should maximize the area of the bounding box of the target. In your opinion, is it feasible to achieve this using AirSIM? Additionally, do you think we can use an object detection algorithm like YOLO in Airsim for target detection?

I don't have experience with AIRSIM and previously worked in the field of deep learning and PyTorch. In your opinion, if I want to customize your code, define my own reward function (EX:area of the bonding box), and set up my environment, which script should I go for?

heleidsn, 2024-06-10:
Hi, from my point of view, your proposed method is feasible in AirSim, as it has an easy way to get images and send control commands to a quadrotor. You can start with `train.py` or `start_train_with_plot.py`. As for the reward function, I think it's better to keep the percentage of the YOLO box at a specific size, such as 1/9 (rather than maximize it as you said). Another important thing is to keep the target in the middle of FOV. You can find the reward function in my code easily.

heleidsn, 2024-06-10:
By the way, the  train.py or start_train_with_plot.py are in the script folder.

khosrooo, 2024-06-10:
Thanku so much
u are the best

heleidsn, 2024-06-11:
Thanks. I will close this issue. If you have any other questions please feel free to create a new one. 

Best

closed 2024-06-11 by heleidsn