heleidsn/UAV_Navigation_DRL_AirSim #38: Problems running start_train_with_plot.py

niguanfeixian, 2024-06-10:
`"D:\Program Installation\Anaconda\envs\Airsim_DRL\python.exe" D:\PycharmProjects\Airsim_DRL\scripts\start_train_with_plot.py 
init training thread
Traceback (most recent call last):
  File "D:\PycharmProjects\Airsim_DRL\scripts\start_train_with_plot.py", line 53, in <module>
    main()
  File "D:\PycharmProjects\Airsim_DRL\scripts\start_train_with_plot.py", line 35, in main
    training_thread = TrainingThread(config_file)
  File "D:\PycharmProjects\Airsim_DRL\scripts\utils\thread_train.py", line 55, in __init__
    self.env.set_config(self.cfg)
  File "D:\PycharmProjects\Airsim_DRL\gym_env\gym_env\envs\airsim_env.py", line 64, in set_config
    self.dynamic_model = MultirotorDynamicsAirsim(cfg)
  File "D:\PycharmProjects\Airsim_DRL\gym_env\gym_env\envs\dynamics\multirotor_airsim.py", line 22, in __init__
    self.client.confirmConnection()
  File "D:\Program Installation\Anaconda\envs\Airsim_DRL\lib\site-packages\airsim\client.py", line 139, in confirmConnection
    if self.ping():
  File "D:\Program Installation\Anaconda\envs\Airsim_DRL\lib\site-packages\airsim\client.py", line 35, in ping
    return self.client.call('ping')
  File "D:\Program Installation\Anaconda\envs\Airsim_DRL\lib\site-packages\msgpackrpc\session.py", line 41, in call
    return self.send_request(method, args).get()
  File "D:\Program Installation\Anaconda\envs\Airsim_DRL\lib\site-packages\msgpackrpc\session.py", line 51, in send_request
    self._transport.send_message([message.REQUEST, msgid, method, args])
  File "D:\Program Installation\Anaconda\envs\Airsim_DRL\lib\site-packages\msgpackrpc\transport\tcp.py", line 89, in send_message
    self.connect()
  File "D:\Program Installation\Anaconda\envs\Airsim_DRL\lib\site-packages\msgpackrpc\transport\tcp.py", line 98, in connect
    socket = ClientSocket(stream, self, self._encodings)
  File "D:\Program Installation\Anaconda\envs\Airsim_DRL\lib\site-packages\msgpackrpc\transport\tcp.py", line 53, in __init__
    BaseSocket.__init__(self, stream, encodings)
  File "D:\Program Installation\Anaconda\envs\Airsim_DRL\lib\site-packages\msgpackrpc\transport\tcp.py", line 12, in __init__
    self._packer = msgpack.Packer(encoding=encodings[0], default=lambda x: x.to_msgpack())
  File "msgpack\\_packer.pyx", line 124, in msgpack._cmsgpack.Packer.__init__
TypeError: __init__() got an unexpected keyword argument 'encoding'
init airsim-gym-env.
Environment:  SimpleAvoid Dynamics:  Multirotor Perception:  depth`
This error has me wondering? Can anyone help with this?

heleidsn, 2024-06-10:
Hi, it seems an error from the environment config file. Please check it in the config folder.

niguanfeixian, 2024-06-10:
After installing the project, it reported an error as soon as it started to run, as shown in Figure 1; then, I changed the address of the configuration file, as shown in Figure 2, and reported this error above.
[image]
[image]

heleidsn, 2024-06-11:
It works on my PC. Please check: 
1) using the latest code
2) debugging with breakpoints to check if you get correct `config_Maze_SimpleMultirotor_2D` config file

ding15963, 2024-11-03:
Did you find a solution for this?

virtue688, 2024-11-05:
> After installing the project, it reported an error as soon as it started to run, as shown in Figure 1; then, I changed the address of the configuration file, as shown in Figure 2, and reported this error above. ![1](https://private-user-images.githubusercontent.com/125891761/338202556-3286f720-9b4e-4ca1-a332-90020c988d81.PNG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzA3ODU2NzAsIm5iZiI6MTczMDc4NTM3MCwicGF0aCI6Ii8xMjU4OTE3NjEvMzM4MjAyNTU2LTMyODZmNzIwLTliNGUtNGNhMS1hMzMyLTkwMDIwYzk4OGQ4MS5QTkc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMTA1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTEwNVQwNTQyNTBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT00MzQ0NjkxZmI4M2JhZWY2MzhkZDMyYjU3NTMyYzQzYWZiY2NhMWU0NzcwYTdiM2MxODA5NDM3YzlhYzAxNGM0JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.nYaBFrbIFz8QWWwsiTpZd2uyIX2ic5-lGX5IwimrLMg) ![2](https://private-user-images.githubusercontent.com/125891761/338202594-8888d4f6-4b10-4511-81a0-c728a4bb9f6b.PNG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzA3ODU2NzAsIm5iZiI6MTczMDc4NTM3MCwicGF0aCI6Ii8xMjU4OTE3NjEvMzM4MjAyNTk0LTg4ODhkNGY2LTRiMTAtNDUxMS04MWEwLWM3MjhhNGJiOWY2Yi5QTkc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMTA1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTEwNVQwNTQyNTBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0yNDBlYjcxNGQzNTExMTZlNTllYmYyZGFkMWIyMGY2NTJhZjA0ZGNlNDdhNWM4ZjBkNzQ5NDMwMzE4MmZjYzNhJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.SmAoGfHWWNCcOjyee38b96k_XxD9kUuNxSc0iWL-Wx0)

do you find some solutions to solve this ?

virtue688, 2024-11-06:
happy!! I have solved this problem, msgpack library's version must <1.0, I use the 0.6.2 ,it works!