heleidsn/UAV_Navigation_DRL_AirSim #39: cannot import name 'spaces' from 'gym'

jinjikikko976, 2024-08-01:
[image]

heleidsn, 2024-08-03:
Hi, please check your gym version as it is an old project, the latest gym version may not work. V0.21.0 should be ok

jinjikikko976, 2024-08-05:
Thank you for your reply. This issue has been resolved, is it still a problem with the environment configuration.But now I have another problem. According to your training settings, the success rate after training in the SimpleAvoid environment is very low, even 0,I don't know how to adjust it

jinjikikko976, 2024-08-05:
[image]

heleidsn, 2024-08-06:
It depends on your configuration, such as perception method, algo selection and settings. Maybe you can upload the config file you use or try to train in different policy and algos

jinjikikko976, 2024-08-06:
I used the default configuration you uploaded and replaced SAC,TD3 and PPO three algorithms, and their success rates were all 0

closed 2024-08-23 by jinjikikko976