heleidsn/UAV_Navigation_DRL_AirSim #40: Env help

jinjikikko976, 2024-09-12:
Could you provide the environment of NH-tree, I can't find it on the official website and I need it, I will be very grateful for your help!

heleidsn, 2024-09-12:
Hi, the NH-tree is the same as AirsimNH, just change the location to somewhere in the forest ^_^

jinjikikko976, 2024-09-12:
> 嗨，NH 树与 AirsimNH 相同，只需将位置更改为森林^_^中的某个位置

Thank you for your reply, I use the compiled NH environment, and it will report an error when  open it with UE for editing, so I want to ask about your environment, if it's convenient for you

heleidsn, 2024-09-12:
You can download the release version from [AirSim release](https://github.com/microsoft/AirSim/releases/tag/v1.6.0-windows). Please note, the code currently only works with AirSim v1.6.0. Download the [AirSimNH.zip](https://github.com/microsoft/AirSim/releases/download/v1.6.0-windows/AirSimNH.zip) and enjoy~

jinjikikko976, 2024-09-13:
> 您可以从 [AirSim 版本](https://github.com/microsoft/AirSim/releases/tag/v1.6.0-windows)下载发布版本。请注意，该代码目前仅适用于 AirSim v1.6.0。下载[AirSimNH.zip](https://github.com/microsoft/AirSim/releases/download/v1.6.0-windows/AirSimNH.zip)享受吧~

Thanks for your reply

closed 2024-09-13 by jinjikikko976