heleidsn/UAV_Navigation_DRL_AirSim #41: Training duration issue

jinjikikko976, 2024-09-27:
Hello, excuse me！May I ask how long it would take for a complete training session.
It usually takes more than 10 hours to train in a 3D environment with 4090,Is there not only a hardware issue but also a code problem that causes this？
Another question is what network and parameter configuration do you use for the experiment on the physical machine？
Your project has brought us great help! We look forward to your reply!

heleidsn, 2024-09-27:
Hi, it depends on the NN configuration. For the simplest case, such as mlp policy in Simple Avoid env, it should be converged around 20k timesteps, which usually takes less than 2 hours on 3060 gpu.  It will take more time for other policies especially the mobile_net one. You can use time funtion in python to see which part cost time most, training or gathering data. 

Yes, the trained parameter can be applied directly to real drone. Jetson Nano is enough to run the network.

jinjikikko976, 2024-09-29:
> Hi, it depends on the NN configuration. For the simplest case, such as mlp policy in Simple Avoid env, it should be converged around 20k timesteps, which usually takes less than 2 hours on 3060 gpu. [...]

Thank you for your reply，I will try using the time function