heleidsn/UAV_Navigation_DRL_AirSim #45: AssertionError: The environment must specify an action space.

SuperheroZLP, 2024-10-23:
D:\software\Anaconda\Anaconda\envs\DRL\python.exe D:\software\AirSim\UAV_Navigation_DRL_AirSim\scripts\start_evaluate_with_plot.py 
init training thread
Traceback (most recent call last):
  File "D:\software\AirSim\UAV_Navigation_DRL_AirSim\scripts\start_evaluate_with_plot.py", line 64, in <module>
    main()
  File "D:\software\AirSim\UAV_Navigation_DRL_AirSim\scripts\start_evaluate_with_plot.py", line 42, in main
    evaluate_thread = EvaluateThread(
  File "D:\software\AirSim\UAV_Navigation_DRL_AirSim\scripts\utils\thread_evaluation.py", line 65, in __init__
    self.env = gym.make('airsim-env-v0')
  File "D:\software\Anaconda\Anaconda\envs\DRL\lib\site-packages\gym\envs\registration.py", line 669, in make
    env = PassiveEnvChecker(env)
  File "D:\software\Anaconda\Anaconda\envs\DRL\lib\site-packages\gym\wrappers\env_checker.py", line 20, in __init__
    assert hasattr(
AssertionError: The environment must specify an action space. https://www.gymlibrary.dev/content/environment_creation/
init airsim-gym-env.

大佬帮帮我，这是怎么回事啊

SuperheroZLP, 2024-10-23:
D:\software\Anaconda\Anaconda\envs\DRL\python.exe D:\software\AirSim\UAV_Navigation_DRL_AirSim\scripts\start_evaluate_with_plot.py 
init training thread
init airsim-gym-env.
Traceback (most recent call last):
  File "D:\software\AirSim\UAV_Navigation_DRL_AirSim\scripts\start_evaluate_with_plot.py", line 64, in <module>
    main()
  File "D:\software\AirSim\UAV_Navigation_DRL_AirSim\scripts\start_evaluate_with_plot.py", line 42, in main
    evaluate_thread = EvaluateThread(
  File "D:\software\AirSim\UAV_Navigation_DRL_AirSim\scripts\utils\thread_evaluation.py", line 65, in __init__
    self.env = gym.make('airsim-env-v0')
  File "D:\software\Anaconda\Anaconda\envs\DRL\lib\site-packages\gym\envs\registration.py", line 669, in make
    env = PassiveEnvChecker(env)
  File "D:\software\Anaconda\Anaconda\envs\DRL\lib\site-packages\gym\wrappers\env_checker.py", line 20, in __init__
    assert hasattr(
AssertionError: The environment must specify an action space.

怎么回事啊这个

JHIN-XIX, 2025-01-09:
请问解决了吗，我遇到了和你一样的错误AssertionError: The environment must specify an action space
[image]

Gxrui, 2025-04-03:
我也遇到了相同的问题，求稳大家解决了吗