heleidsn/UAV_Navigation_DRL_AirSim #50: what kind of network  in  airsimNH is useful?

virtue688, 2024-11-26:
Hello, I have a question to ask. In a complex environment like AirSimNH, the navigation success rate is very low when using No_CNN for training. I am considering switching to other networks or adding an attention module. What kind of network do you think would perform better in such a complex environment? Or have you tried any networks that work particularly well in the AirSimNH environment?

heleidsn, 2024-11-28:
Hi, good question. I tried some networks, but they are all based on the current states, which can be considered as a reactive planner (or controller). Actually, for the AirSimNH envrionment, the simple one No_cnn can still achieve quite good success rate, about 70% I remember. Maybe adding attention module and uisng time sequtial input will be better for the complex environment, but they need more data and time for the training process.

virtue688, 2024-11-29:
Thank you very much for your answer. I would also like to ask you, why do you think convolution performs poorly in the depth map feature extraction network? Are the features of the depth map too simple? Have you ever tried the effect of using a cnn network in an actor network? Thanks again for your reply

heleidsn, 2024-12-02:
CNN is good to extract features more related to edges. For obstacle avoidance, the position of obstacles is better than eage features, that's why I don't recommand to use CNN in depth input for obstacle avoidance problem. As it will increase the training difficulty but cannot bring us more useful information. (only from my point of view, we can disscuss more if you have other opinion)

virtue688, 2024-12-03:
Thank you very much! In fact, I have tried some attention modules in feature_extract network, but it doesn't work (about 40% accuracy).  I found that if the goal point in the deeper position in the airsimNH map, the UAV can't arrive at the goal, so I think it may be that you are faced with complex obstacles or you are unable to complete obstacle avoidance. Does this success rate need to be achieved by adjusting parameters, such as adjusting the learning rate and other parameters?
[image]

heleidsn, 2024-12-03:
Thanks for your reply. The success rate should be imporved to about 80% with the simple no_cnn policy if I remember correctly. For long waypoint task, you can consider the controller as a reactive controller, it only guide the drone to turn left or turn right or climb to avoid the obstacle. Maybe you can still try the simple policy and improve it when neccesary. Best

virtue688, 2024-12-03:
Thank for your reply! If you use the velocity state,please?

heleidsn, 2024-12-03:
Velocity information is useful, but not compulsory I think. For my training, I didn't use the velocity state finally, I just consider the policy as a reactive controller to generate velocity setpoint command and assume the low level velocity tracking is perfect without control error. You can compare these two methods with and without velocity state to see which one is better.

virtue688, 2024-12-03:
Thank you very much! I will have a try!

virtue688, 2025-03-25:
> 请问您这些成功率碰撞率的图是怎么生成出来的

用wannb，在config文件中有是否使用wannb选项

Flieluo, 2025-09-19:
> 您好，请问从2d到NH的3d您做了哪些配置呢，我2d成功了，NH不能成功

请问您是否配置出了适用于NH的3d的参数呢

Flieluo, 2025-09-19:
> 速度信息很有用，但我认为不是强制性的。对于我的训练，我最终没有使用速度状态，我只是将策略视为生成速度设定点命令的反应式控制器，并假设低级速度跟踪是完美的，没有控制错误。您可以比较这两种有和没有速度状态的方法，看看哪一种更好。

Could you share the parameter and config files compatible with the NH 3D environment?

Flieluo, 2025-09-19:
> > 请问您这些成功率碰撞率的图是怎么生成出来的
> 
> 用wannb，在config文件中有是否使用wannb选项
Could you share the parameter and config files compatible with the NH 3D environment?

WuJiarui-cwnu, 2025-11-09:
> Thank you very much! I will have a try!非常感谢！我会试试看！

Could you kindly share your paper? Thank you very much.

virtue688, 2025-11-17:
> > Thank you very much! I will have a try!非常感谢！我会试试看！
> 
> Could you kindly share your paper? Thank you very much.

I plan to submit my paper the next year，IROS 2026，i will share paper after it

Exclusived, 2025-11-18:
> > > 非常感谢！我会尝试一下！非常感谢！我会尝试看看！
> > 
> > 
> > 能否请您分享一下您的论文？非常感谢。
> 
> 我计划明年提交论文，即 IROS 2026，之后我会分享论文。
@virtue688 
这里面的SAC,TD3和PPO哪个算法训练的效果最好

HY-0928, 2026-02-08:
> > > Thank you very much! I will have a try!非常感谢！我会试试看！
> > 
> > 
> > Could you kindly share your paper? Thank you very much.
> 
> I plan to submit my paper the next year，IROS 2026，i will share paper after it

您好！我想请教一个问题，我在AirsimNH这种场景下训练智能体时，无人机会出现在起始点就碰到障碍物（树木），然后就一直卡在这个位置的情况，并且无人机在飞行过程中十分抖动，不知道这个是不是我的配置出了问题。您遇到过上述问题吗？或者可以分享一下您在这个场景下的配置文件的相关变动吗？十分感谢！

Flieluo, 2026-02-08:
大概什么时间论文会见刊啊

---- Replied Message ----
| From | ***@***.***> |
| Date | 02/08/2026 19:23 |
| To | heleidsn/UAV_Navigation_DRL_AirSim ***@***.***> |
| Cc | Flieluo ***@***.***>,
Comment ***@***.***> |
| Subject | Re: [heleidsn/UAV_Navigation_DRL_AirSim] what kind of network in airsimNH is useful? (Issue #50) |
HY-0928 left a comment (heleidsn/UAV_Navigation_DRL_AirSim#50)

Thank you very much! I will have a try!非常感谢！我会试试看！

Could you kindly share your paper? Thank you very much.

I plan to submit my paper the next year，IROS 2026，i will share paper after it

您好！我想请教一个问题，我在AirsimNH这种场景下训练智能体时，无人机会出现在起始点就碰到障碍物（树木），然后就一直卡在这个位置的情况，并且无人机在飞行过程中十分抖动，不知道这个是不是我的配置出了问题。您遇到过上述问题吗？或者可以分享一下您在这个场景下的配置文件的相关变动吗？十分感谢！

virtue688, 2026-02-09:
我尝试了SAC和TD3效果都很不错，我没有碰到卡住的情况但无人机飞行中的抖动是正常的。wannb现在大陆似乎用不了了，我已经没有使用wannb了。论文目前还在审稿中

Flieluo, 2026-02-10:
请问PPO可以用吗

---- Replied Message ----
| From | ***@***.***> |
| Date | 02/09/2026 12:27 |
| To | heleidsn/UAV_Navigation_DRL_AirSim ***@***.***> |
| Cc | Flieluo ***@***.***>,
Comment ***@***.***> |
| Subject | Re: [heleidsn/UAV_Navigation_DRL_AirSim] what kind of network in airsimNH is useful? (Issue #50) |
virtue688 left a comment (heleidsn/UAV_Navigation_DRL_AirSim#50)

我尝试了SAC和TD3效果都很不错，我没有碰到卡住的情况但无人机飞行中的抖动是正常的。wannb现在大陆似乎用不了了，我已经没有使用wannb了。论文目前还在审稿中