heleidsn/UAV_Navigation_DRL_AirSim #51: 为什么训练不出来效果呢，飞机一直撞墙呢

datutu-L, 2024-12-12:
[image]
请教一下为什么训练完的效果特别不好呢，我没修改代码中的任何参数，飞机一直在撞墙。
有大佬能帮忙解答一下吗。

datutu-L, 2024-12-13:
您好，请问参数该如何设置呢，就按照仓库中文件[config_Maze_SimpleMultirotor_2D.ini](https://github.com/heleidsn/UAV_Navigation_DRL_AirSim/blob/main/configs/config_Maze_SimpleMultirotor_2D.ini)里面的默认参数就可以训练出来吗 :sob: @heleidsn

virtue688, 2024-12-27:
> 您好，请问参数该如何设置呢，就按照仓库中文件[config_Maze_SimpleMultirotor_2D.ini](https://github.com/heleidsn/UAV_Navigation_DRL_AirSim/blob/main/configs/config_Maze_SimpleMultirotor_2D.ini)里面的默认参数就可以训练出来吗 😭 @heleidsn

NOCNN效果比较好

15553117127, 2024-12-30:
我也这样，用训练出的模型也一直撞墙

heleidsn, 2024-12-30:
不好意思，现在手边没有配置好的环境来验证，一直撞墙的话可能有两种原因：
1. 刚开始有一段时间是随机探索阶段，用来给replay buffer填一些数据，然后才会开始训练
2. 可以查看一下action的noise设置是否合适

总体来说用NO-CNN的policy和SAC算法应该很快就会收敛的

wangyi11232, 2025-03-07:
您解决了吗  我也一直撞墙用的mlp

closed 2025-12-10 by datutu-L