heleidsn/UAV_Navigation_DRL_AirSim #64: 作者大大，使用config_SimpleMultirotor_2D.ini配置文件进行训练，为啥到训练结束成功率一直为0呢

ytlz7, 2025-04-07:


heleidsn, 2025-04-10:
你好，我也很久没有运行这个项目了，一般成功率低可能是初始化网络参数不太好，然后训练陷入了local minimum，可以尝试增加action noize的大小看看会不会改善。此外，我自己的经验，用SAC会比TD3训练快一些。

w4646, 2025-05-07:
> 你好，我也很久没有运行这个项目了，一般成功率低可能是初始化网络参数不太好，然后训练陷入了local minimum，可以尝试增加action noize的大小看看会不会改善。此外，我自己的经验，用SAC会比TD3训练快作者大大你好，那个resources/results中的那几张图是怎么生成的呢？

heleidsn, 2025-05-08:
> > 你好，我也很久没有运行这个项目了，一般成功率低可能是初始化网络参数不太好，然后训练陷入了local minimum，可以尝试增加action noize的大小看看会不会改善。此外，我自己的经验，用SAC会比TD3训练快作者大大你好，那个resources/results中的那几张图是怎么生成的呢？

那个图是用多种算法训练之后，自己拿python画的，用seaborn

w4646, 2025-07-05:
> > > 你好，我也很久没有运行这个项目了，一般成功率低可能是初始化网络参数不太好，然后训练陷入了local minimum，可以尝试增加action noize的大小看看会不会改善。此外，我自己的经验，用SAC会比TD3训练快作者大大你好，那个resources/results中的那几张图是怎么生成的呢？
> 
> 那个图是用多种算法训练之后，自己拿python画的，用seaborn

你好，像Tree等这些环境在哪下载呀