heleidsn/UAV_Navigation_DRL_AirSim #67: 请问有大佬尝试过用CNN_GAP等CNN策略训练出不错的模型吗？

AdaptZZC, 2025-05-10:
我利用No_CNN策略和SAC算法可以训练出一个不错的模型，但是当策略变成CNN策略时，模型总是不能收敛，无人机一直撞墙。根据作者Readme中的数据图SAC的CNN_GAP策略大约十万步将会有不错的训练效果，但我目前的参数并没有较好的结果

请问有大佬可以指点一下吗？或者可以交流一下，非常感谢

[image]

AdaptZZC, 2025-09-24:
> > 我利用No_CNN策略和SAC算法可以训练出一个不错的模型，但是当策略变成CNN策略时，模型总是不能收敛，无人机一直撞墙。根据作者Readme中的数据图SAC的CNN_GAP策略大约十万步将会有不错的训练效果，但我目前的参数并没有较好的结果
> > 请问有大佬可以指点一下吗？或者可以交流一下，非常感谢
> > [image]
> 
> 请问你有尝试过env_name为NH_center环境下， 跑出一个好的效果吗
没有，我只尝试了Simple那个环境

datutu-L, 2025-12-10:
> > 我利用No_CNN策略和SAC算法可以训练出一个不错的模型，但是当策略变成CNN策略时，模型总是不能收敛，无人机一直撞墙。根据作者Readme中的数据图SAC的CNN_GAP策略大约十万步将会有不错的训练效果，但我目前的参数并没有较好的结果
> > 请问有大佬可以指点一下吗？或者可以交流一下，非常感谢
> > [image]
> 
> 请问你有尝试过env_name为NH_center环境下， 跑出一个好的效果吗

请问NH_center环境下你训练出来了吗